Dr-Gigavolt/dbus-aggregate-batteries #82: Issue with “Aggregate Batteries” Not Appearing in GUI as Device

thenebu, 2025-01-17:
Hi there,

I am using two battery packs, both integrated with SerialBattery. In the GUI and dbus-spy, I can see both battery packs correctly listed. However, I am unable to see the device “Aggregate Batteries” in the GUI.

In Settings → System Setup → Batteries, I can select “AggregateBatteries on Virtual” as the battery monitor. However, there is no corresponding device, and the log file indicates that no batteries are being detected.

Here’s the relevant portion of the log:
```
root@einstein:~# tail -F /var/log/dbus-aggregate-batteries/current
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi0_uc272427 has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyS7, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyS7 has device instance 279
INFO:dbusmonitor:Found: com.victronenergy.system, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.system has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyS4 has device instance 276
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Fri Jan 17 09:09:21 2025: Searching batteries: Trial Nr. 1
INFO:root:Fri Jan 17 09:09:21 2025: 0 batteries found.
INFO:root:Fri Jan 17 09:09:26 2025: Searching batteries: Trial Nr. 2
INFO:root:Fri Jan 17 09:09:26 2025: 0 batteries found.
INFO:root:Fri Jan 17 09:09:31 2025: Searching batteries: Trial Nr. 3
INFO:root:Fri Jan 17 09:09:31 2025: 0 batteries found.
INFO:root:Fri Jan 17 09:09:36 2025: Searching batteries: Trial Nr. 4
INFO:root:Fri Jan 17 09:09:36 2025: 0 batteries found.
```

As you can see, it appears that the system is unable to detect the batteries, even though they are visible on dbus-spy.

Could you please advise on what might be causing this issue and how I can resolve it?

I’ve also attached a screenshot for reference:

[image]

Dr-Gigavolt, 2025-01-19:
Hello, the "AggregateBatteries on Virtual" is correct:

[image]

Your log says that the instance AggregateBatteries was created, but only one battery was found by dbusmonitor. And this one was not recognized by my program. Please attach your settings.py and dbus-spy screenshot of both batteries.

Here is my log after restart: 

INFO:root:Sun Jan 19 19:25:34 2025: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:Sun Jan 19 19:25:34 2025: Initial Ah read from file: 118Ah
INFO:root:Sun Jan 19 19:25:34 2025: Last balancing done at the 302. day of the year
INFO:root:Batteries balanced 82 days ago.
INFO:root:Sun Jan 19 19:25:34 2025: Starting battery monitor.
INFO:root:Sun Jan 19 19:25:34 2025: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 1
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB4 has device instance 3
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.system, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.system has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyUSB1 has device instance 290
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB0 has device instance 288
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Sun Jan 19 19:25:35 2025: Searching Settings: Trial Nr. 1
INFO:root:Sun Jan 19 19:25:35 2025: com.victronenergy.settings found.
INFO:root:Sun Jan 19 19:25:40 2025: Searching batteries: Trial Nr. 1
INFO:root:Sun Jan 19 19:25:40 2025: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms).
INFO:root:Sun Jan 19 19:25:40 2025: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms)2.
INFO:root:Sun Jan 19 19:25:40 2025: 2 batteries found.
INFO:root:Sun Jan 19 19:25:41 2025: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:Sun Jan 19 19:25:41 2025: MultiPlus-II 48/5000/70-50 found.
INFO:root:Sun Jan 19 19:25:42 2025: Searching MPPTs: Trial Nr. 1
INFO:root:Sun Jan 19 19:25:42 2025: SmartSolar MPPT VE.Can 250/100 rev2 found.
INFO:root:Sun Jan 19 19:25:42 2025: 1 MPPT(s) found.
INFO:root:Sun Jan 19 19:25:43 2025: CVL increase for balancing activated.

Mark21041981, 2025-01-24:
How did you connect the batteries with what cables? I only see one batery nd i have three of them.

Do you have a link for the cable?

Dr-Gigavolt, 2025-01-26:
Hello Mark, do you ask me, or thenebu? I connected the 3.3V serial interfaces of my two JK BMS by screened cables (screen = GND) to insulated serial/USB adaptors (https://www.ebay.de/itm/265430429754). These are plugged into a USB hab connected with Raspberry Pi.

thenebu, 2025-02-09:
Whats this about?
WARNING:root:USING OUTDATED REGISTRATION METHOD! WARNING:root:Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api 

I used Battery Aggregator the last weeks and it worked. But I wasn't happy with the CCL Calculation, so I wanted to try your Aggregate Batteries again... still some issues as you can see

thenebu, 2025-02-09:
With some sleep I got it. Sorry

edited:
`BATTERY_PRODUCT_NAME = "SerialBattery" to BATTERY_PRODUCT_NAME = "BMS"`

and got rid of the "outdated" error of

edited the class DbusAggBatService(object): in aggregatebatteries.py
```
self._dbusservice = VeDbusService(servicename, register=False)
self._dbusservice.register()
```

PS: in settings.py you use # Key **world** to identify instead of "**word**". :)

closed 2025-02-09 by thenebu