Dr-Gigavolt/dbus-aggregate-batteries #83: Error when reading battery details

TheGuru0000, 2025-02-11:
You can find the errors I am referring to at the bottom:

`INFO:root:Tue Feb 11 22:19:00 2025: Starting AggregateBatteries.
WARNING:root:USING OUTDATED REGISTRATION METHOD!
WARNING:root:Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:Tue Feb 11 22:19:00 2025: Initial Ah read from file: 306Ah
INFO:root:Tue Feb 11 22:19:00 2025: Starting battery monitor.
INFO:root:Tue Feb 11 22:19:00 2025: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyS6, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyS6 has device instance 277
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi2_uc305391, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi2_uc305391 has device instance 2
INFO:dbusmonitor:Found: com.victronenergy.system, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.system has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
INFO:root:Tue Feb 11 22:19:01 2025: Searching Settings: Trial Nr. 1
INFO:root:Tue Feb 11 22:19:01 2025: com.victronenergy.settings found.
INFO:dbusmonitor:       com.victronenergy.vebus.ttyS4 has device instance 276
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi0_uc305012, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi0_uc305012 has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0__0x01, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0__0x01 has device instance 1
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0__0x02, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0__0x02 has device instance 2
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi1_uc305194, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi1_uc305194 has device instance 1
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.battery.socketcan_can0_vi0_uc337291, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.socketcan_can0_vi0_uc337291 has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Tue Feb 11 22:19:06 2025: Searching batteries: Trial Nr. 1
INFO:root:Tue Feb 11 22:19:06 2025: SerialBattery(JKBMS PB Model) found, named as: SerialBattery JKBMS .
INFO:root:Tue Feb 11 22:19:06 2025: SerialBattery(JKBMS PB Model) found, named as: SerialBattery(JKBMS PB Model).
INFO:root:Tue Feb 11 22:19:06 2025: 2 batteries found.
INFO:root:Tue Feb 11 22:19:07 2025: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:Tue Feb 11 22:19:07 2025: MultiPlus-II 48/5000/70-50 found.
INFO:root:Tue Feb 11 22:19:08 2025: Searching MPPTs: Trial Nr. 1
INFO:root:Tue Feb 11 22:19:08 2025: SmartSolar Charger MPPT 100/20 48V found.
INFO:root:Tue Feb 11 22:19:08 2025: SmartSolar MPPT VE.Can 150/85 rev2 found.
INFO:root:Tue Feb 11 22:19:08 2025: SmartSolar MPPT VE.Can 150/85 rev2 found.
INFO:root:Tue Feb 11 22:19:08 2025: SmartSolar MPPT VE.Can 150/85 rev2 found.
INFO:root:Tue Feb 11 22:19:08 2025: 4 MPPT(s) found.
ERROR:root:Tue Feb 11 22:19:09 2025: Error: Battery SerialBattery JKBMS  returns None value of /Voltages/Sum..
ERROR:root:Occured during step Read max. and min cell voltages and voltage sum, Battery SerialBattery JKBMS .
ERROR:root:Read trial nr. 1
ERROR:root:Tue Feb 11 22:19:10 2025: Error: Battery SerialBattery JKBMS  returns None value of /Voltages/Sum..
ERROR:root:Occured during step Read max. and min cell voltages and voltage sum, Battery SerialBattery JKBMS .
ERROR:root:Read trial nr. 2
ERROR:root:Tue Feb 11 22:19:11 2025: Error: Battery SerialBattery JKBMS  returns None value of /Voltages/Sum..
ERROR:root:Occured during step Read max. and min cell voltages and voltage sum, Battery SerialBattery JKBMS .
ERROR:root:Read trial nr. 3
`
I am using aggregate_batteries ## Version 3.3 and
Venus OS v3.53 
dbus-serialbattery v1.6.20250131

Here is also the - I think- relevant part of the settings.py:
`# ######################################
# ########### DBus settings ############
# ######################################

# Key world to identify services of physical Serial Batteries (and SmartShunt if available)
BATTERY_SERVICE_NAME = "com.victronenergy.battery.ttyUSB"
# Path of Battery Product Name
BATTERY_PRODUCT_NAME_PATH = "/ProductName"
# Key world to identify the batteries (to exclude SmartShunt)
BATTERY_PRODUCT_NAME = "SerialBattery"
# if CustomName doesn't exist, set '/ProductName'
BATTERY_INSTANCE_NAME_PATH = "/Serial"
# Key world to identify service of Multis/Quattros (or cluster of them)
MULTI_KEY_WORD = "com.victronenergy.vebus"
# Key world to identify services of solar chargers
MPPT_KEY_WORD = "com.victronenergy.solarcharger"
# Key world to identify services of SmartShunt
SMARTSHUNT_NAME_KEY_WORD = "Lynx Shunt"
`
dbus-spy gives me these details for one of the batteries:

[image]

Did I configure something obviously wrong to trigger the error?

TheGuru0000, 2025-02-13:
I did some further analysis and looked into how the dbus basically works.
My conclusion is that the latest serial_battery driver does not offer the required dbus information ("/Voltages/Sum") any more.
That means, that the aggregate_batteries in the version 3.3 will not work with current serial_battery-driver versions any more. 

However, the change is quite simple.
I did a fork of the project and will upload my changes, that I use to run the aggregate_batteries successfully.
@Dr-Gigavolt: I can put a small pull-request for Your repo, if You are ok with that? 

I also saw, that the aggregate_batteries expects more information for the cells under /Voltages/, and the whole /Voltages/ area is not offered by the serial_battery driver (any more). Luckily aggregate_batteries seems to not need/use these other information, but I did not have the time to check details.

In the process of doing this, i also found out, that my Lynx Shunt conflicts with the aggregate_batteries drivers - i cannot see both at the same time in the battery-driver, that the cerbo uses. When I want to use aggregate_batteries, i need to unplug the shunt from the Ve.can connection.

Dr-Gigavolt, 2025-02-13:
Hi and thanks for the reporting. I'm in a terrible lack of time and not able to follow changes in Serial Battery. You can make the pull request, but please think about that in the code there are already some unreleased changes. And think about the backwards compatibility. I have no time to test it. Thanks.

madtv, 2025-02-21:
> Hello, it is nice that the description says. I have never promised it. It is same as in your case, I have no time and sometimes I think about retiring this project. I wrote the tool for myself and published it for others, but I cannot fulfil every wish. I make the new release now (I added some more diagnostics) and you can add another data format if you think it is important. But please test it properly before uploading. About Lynx shunt - do you really need it? My tool collects current measurements from all Multis/Quatros/Chargers and these are very precise.

Dear Anton, thank you for the update! It works so far and please do not stop the development. You have a very large fan community who need your project. You are doing a really great job! Thank you!

closed 2025-03-05 by TheGuru0000