Dr-Gigavolt/dbus-aggregate-batteries #94: When "OWN_CHARGE_PARAMETERS = False".

River-Phoenix-13, 2025-05-08:
I have 3x LiFePo4 200Ah, 25.6V, with Daly BMS batteries in series. 
If I set OWN_CHARGE_PARAMETERS = False then after 1 battery is going into float in serial-battery. Aggregate-Batteries will then use the float parameters from Serial-Batteries. That means that charging from then on will be done with the Float parameters and the remaining batteries are not charged to full.
Is the above normal behavior or is intended that "bulk" should be used until all batteries are in float?

Dr-Gigavolt, 2025-05-08:
It is normal safe behavior. It can be changed, please read the comments in settings.py carefully.

River-Phoenix-13, 2025-05-08:
I think I read and understand it all.
The only change should be: KEEP_MAX_CVL = True 
I did that and still aggregate-batteries does what I described in my first post.

Dr-Gigavolt, 2025-05-11:
I have never tried it and I even not recommend to use this option. I made it on explicit wish of some Serial Battery users. Up to now I heard no complains, I recommend you to ask that guys about their experience. I cannot review my code immediately, no time.  If you can wait one or two weeks I will have a look.

YetAnotherOneGuru, 2025-05-12:
Hi @Dr-Gigavolt / Anton,   
I also wanted to use the function KEEP_MAX_CVL = True and saw last week, that this switch seems to not have any impact.

I searched in the github repo and saw, that You introduced this after #52, specifically with the comment https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/52#issuecomment-1784154222

However, when checking the responsible code, i only find:

`# find max. charge voltage (if needed)
        if not settings.OWN_CHARGE_PARAMETERS:
            MaxChargeVoltage = self._fn._min(MaxChargeVoltage_list)  # add KEEP_MAX_CVL
            MaxChargeCurrent = (
                self._fn._min(MaxChargeCurrent_list) * settings.NR_OF_BATTERIES
            )
            MaxDischargeCurrent = (
                self._fn._min(MaxDischargeCurrent_list) * settings.NR_OF_BATTERIES
            )
`  

Apart from the added comment "# add KEEP_MAX_CVL" I do not see any related code change to use this switch.

I know, that Your time is limited, but can You maybe have a look and see, if the actual code change got lost before the commit to github?

River-Phoenix-13, 2025-05-12:
I have the time no problem. I just wanted to see which way of charging suites my system the best.

River-Phoenix-13, 2025-05-12:
Yesterday I installed v2.0.20250502dev off Serial-Battery and with that version OWN_CHARGE_PARAMETERS = FALS" is working. It is then using serial-battery charge protocol and keeps charging until all 3 batteries are in float.

YetAnotherOneGuru, 2025-05-13:
> Yesterday I installed v2.0.20250502dev off Serial-Battery and with that version OWN_CHARGE_PARAMETERS = FALS" is working. It is then using serial-battery charge protocol and keeps charging until all 3 batteries are in float.

strange - i also have that setup and it does not work.
Also the code `MaxChargeVoltage = self._fn._**min**(MaxChargeVoltage_list)` does not give much room for interpretation ;-)
Are You sure, You don't have the "longest balancing" battery set as controlling BMS in the DVCC of Your cerbo?

River-Phoenix-13, 2025-05-13:
I double checked and Aggregate battery is the BMS that's controlling DVCC.
If you have the same setup then I don't know why it`s working here after updating to v2.0.20250502dev.

Here is my settings.py:
# ######################################
# ######## Hardware settings ###########
# ######################################

# Nr. of physical batteries to be aggregated. Smart shunt for battery current is not needed and not supported.
NR_OF_BATTERIES = 3
NR_OF_CELLS_PER_BATTERY = 8

# Nr. of MPPTs
NR_OF_MPPTS = 2

# If DC loads with Smart Shunt present, can be used for total current measurement.
DC_LOADS = False

# False: Current subtracted, True: Current added
INVERT_SMARTSHUNT = False

# ######################################
# ########### DBus settings ############
# ######################################

# Key world to identify services of physical Serial Batteries (and SmartShunt if available). You don't need to change it.
BATTERY_SERVICE_NAME = "com.victronenergy.battery"

# Path of Battery Product Name. You don't need to change it.
BATTERY_PRODUCT_NAME_PATH = "/ProductName"

# Key world to identify the batteries (to exclude SmartShunt). If the BATTERY_PRODUCT_NAME_PATH, e.g. SerialBattery(Jkbms),
# contains this key word, the device will be identified and included into the battery list. 
BATTERY_PRODUCT_NAME = "SerialBattery"

# The name stored in the here selected BATTERY_INSTANCE_NAME_PATH will be taken as the name of the battery in the list.
# Each battery instance should have an unique name. If not, a number will be added. You can choose: "/CustomName"
# (set up in the serial battery driver and is lost on restart of the serial battery driver) or "/Serial"
# (set up in the BMS and therefore not volatile).
BATTERY_INSTANCE_NAME_PATH = "/CustomName"

# Key world to identify service of Multis/Quattros (or cluster of them).  You don't need to change it.
MULTI_KEY_WORD = "com.victronenergy.vebus"

# Key world to identify services of solar chargers. You don't need to change it.
MPPT_KEY_WORD = "com.victronenergy.solarcharger"

# Key world to identify services of SmartShunt, if you use it for current into DC loads. You don't need to change it.
SMARTSHUNT_NAME_KEY_WORD = "SmartShunt"

# Trials to identify of all batteries before exit and restart.
SEARCH_TRIALS = 10

# Trials to get consistent data of all batteries before exit and restart.
READ_TRIALS = 10

# ######################################
# ############# Options ################
# ######################################

# If True, the battery current measurement by Multis/Quattros and MPPTs is taken instead of BMS.
# Necessary for JK BMS due to poor current measurement precision.
# The Victron current measurement is very precise, therefore SmartShunt is not needed and not supported.
CURRENT_FROM_VICTRON = False

# If True, the program's own charge counter is used instead of the BMS counters.
# Necessary for JK BMS due to poor current measurement precision
# and not implemented 100% and 0% reset except of case of MOSFET disconnection.
OWN_SOC = False

[... 507 lines cut ...]

; BMS poll interval in seconds.
; If the driver consumes too much CPU, you can increase this value to reduce the refresh rate
; and CPU usage.
; Default for most BMS is 1 second; some BMS may have a higher value.
; Leave empty to use the BMS default value; decimal values are allowed.
POLL_INTERVAL =

; -- Battery Capacity
; Some BMS do not support reading the battery capacity. Specify the battery capacity in Ah.
; This is applicable for:
;     - Daly
;     - Felicity
BATTERY_CAPACITY = 200

; -- Auto Reset BMS SoC
; If enabled, the BMS SoC is reset to 100% when the voltage switches from absorption to float.
; Requires CVCM_ENABLE to be enabled.
; This is applicable for:
;     - Daly BMS
;     - JKBMS BLE
AUTO_RESET_SOC = True

; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = -1

YetAnotherOneGuru, 2025-05-14:
> strange - i also have that setup and it does not work. Also the code `MaxChargeVoltage = self._fn._min(MaxChargeVoltage_list)` does not give much room for interpretation ;-) Are You sure, You don't [...]

I just checked, i dont have the new beta yet, still running on v1.6.20250131 stable release, sorry. So I will also try out the new beta - but this may take a bit, I need to take some time for testing.

Dr-Gigavolt, 2025-05-14:
Hello guys, I had to upgrade everything to the newest state first, it consumed a half of night :-(.  I hope I will manage to check the program in following couple of days.

Dr-Gigavolt, 2025-05-15:
I mean is it OK for you to keep:
            MaxChargeCurrent = (
                self._fn._min(MaxChargeCurrent_list) * settings.NR_OF_BATTERIES
            )
?

Dr-Gigavolt, 2025-05-15:
I've just pushed it into the repo. I will make release later, I want to check and clean up one more thing.

River-Phoenix-13, 2025-06-14:
On my system it`s working fine, all 3 batteries are going into "float" and then the system goes back to the voltage from the CHARGE_VOLTAGE_LIST.
i`m using Aggregate batteries 3.5.20250516 and Serialbattery  2.0.202518dev

River-Phoenix-13, 2025-06-16:
Copyed the newest aggregatebatteries.py and charged today. 
When ["OWN_CHARGE_PARAMETERS = False"],. All 3 batteries are going one by one in float mode and until all 3 where in float. The charged voltage was 28V max 3,50V cell voltage, in my case the last battery took about 2 hours longer to get balanced and then go to float. So, for me it`s working fine.

River-Phoenix-13, 2025-06-17:
This morning the SOC was 99% and should be 91% and voltage reading 26.9V and should be 26.59. I tought maybe I need to set DC_LOADS = True but then AggregateBatteries will not run see the log.
LOG:
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.digitalinputs
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.modbusclient.tcp
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.temperature.ruuvi_e7462fcd2567
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.temperature.ruuvi_c56d59a480ed
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.battery.ttyS5
INFO:root:Tue Jun 17 04:45:23 2025: Correct Smart Shunt product name SmartShunt 500A/50mV found in the service com.victronenergy.battery.ttyS5
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.ble
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.logger
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.battery.aggregate
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.adc
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.tank.adc_builtin0_1
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.modbustcp
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.tank.adc_builtin0_2
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.battery.ttyUSB0
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.tank.adc_builtin0_3
INFO:root:Tue Jun 17 04:45:23 2025: Dbusmonitor sees: com.victronenergy.tank.adc_builtin0_4
INFO:root:Tue Jun 17 04:45:23 2025: 0 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyACM2 has device instance 4
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 2
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 3
INFO:root:Tue Jun 17 04:45:28 2025: Searching batteries: Trial Nr. 7
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.settings
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.solarcharger.ttyS6
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.solarcharger.ttyS7
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.temperature.ruuvi_cc166fc80d50
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.vecan.can0
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.system
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.fronius
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.battery.ttyACM0
INFO:root:Tue Jun 17 04:45:28 2025: Correct battery product name SerialBattery(Daly) found in the service com.victronenergy.battery.ttyACM0
INFO:root:Tue Jun 17 04:45:28 2025: SerialBattery(Daly) found, named as: SB-2_M.
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.vebus.ttyS4
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.tank.adc_gxtank_HQ2214Q9P2C_0
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.battery.ttyACM1
INFO:root:Tue Jun 17 04:45:28 2025: Correct battery product name SerialBattery(Daly) found in the service com.victronenergy.battery.ttyACM1
INFO:root:Tue Jun 17 04:45:28 2025: SerialBattery(Daly) found, named as: SB-1_BB.
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.tank.adc_gxtank_HQ2214Q9P2C_1
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.battery.ttyACM2
INFO:root:Tue Jun 17 04:45:28 2025: Correct battery product name SerialBattery(Daly) found in the service com.victronenergy.battery.ttyACM2
INFO:root:Tue Jun 17 04:45:28 2025: SerialBattery(Daly) found, named as: SB-3_SB.
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.temperature.ruuvi_f8eb27de548c
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.platform
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.tank.adc_gxtank_HQ2214Q9P2C_2
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.tank.adc_gxtank_HQ2214Q9P2C_3
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.digitalinputs
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.modbusclient.tcp
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.temperature.ruuvi_e7462fcd2567
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.temperature.ruuvi_c56d59a480ed
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.battery.ttyS5
INFO:root:Tue Jun 17 04:45:28 2025: Correct Smart Shunt product name SmartShunt 500A/50mV found in the service com.victronenergy.battery.ttyS5
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.ble
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.logger
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.battery.aggregate
INFO:root:Tue Jun 17 04:45:28 2025: Dbusmonitor sees: com.victronenergy.adc
[... cut ...]
INFO:root:  CVL: 26.9V, CCL: 225A, DCL: 450A
INFO:root:  Bat. voltage: 26.6V, Bat. current: 0A, SoC: 89.6%, Balancing state: 0
INFO:root:  Min. cell voltage: SB-3_SB_C1: 3.322V, Max. cell voltage: SB-2_M_C3: 3.331V, difference: 0.009V
INFO:root:Tue Jun 17 05:05:37 2025: Repetitive logging:
INFO:root:  CVL: 26.9V, CCL: 225A, DCL: 450A
INFO:root:  Bat. voltage: 26.6V, Bat. current: -0A, SoC: 89.6%, Balancing state: 0
INFO:root:  Min. cell voltage: SB-3_SB_C1: 3.322V, Max. cell voltage: SB-2_M_C3: 3.331V, difference: 0.009V

River-Phoenix-13, 2025-06-17:
The SOC and Voltage from AggregateBatteries should be the average of the 3 SerialBattery batteries. And that was not showing it, it was of by 9% and 0.3V. i don`t know why. It happend this afternoon again with the SOC, I corrected that in the "charge" file.
I will keep monitoring it and see if it happens again.
I don't use current or DC loads by Victron shunt, just the current from the daly's. But I can check with my victron shunt if the Daly`s are doing the same.

River-Phoenix-13, 2025-06-18:
For now everything is still running fine, no problems and correct numbers. Maybe the problem in my previous post arrised because the Cerbo is running against it limits.