Dr-Gigavolt/dbus-aggregate-batteries #95: [question] Combine battery brands

petrklus, 2025-05-10:
I am curently planning to add a DIY battery to my existing BYD LVL battery system. The DIY battery will use JK BMS and will be reported in my Cerbo.

Can I use this aggregator to aggreate the JK batteries (multiple) with my existing BYD battery? Both battery types have same number of cells etc.

Dr-Gigavolt, 2025-05-11:
I'm not the right to answer this question. You have to check if the Serial Battery can serve different BMSes. If yes and all instances provide valid data it should work.

petrklus, 2025-05-11:
Thank you for the clarification - the other battery is reported into my Victron system, would it not be possible to read its values without using serial battery? Or is this project focussing on BMSs that use serial battery backend?

Dr-Gigavolt, 2025-05-11:
How does the BYD battery communicate with Victron? Can Bus and no use of Serial Battery driver? Please open a terminal and type dbus-spy command. Navigate with arrows, find your battery, go there and make a screenshots of all published data.

petrklus, 2025-05-12:
@Dr-Gigavolt @YetAnotherOneGuru makes complete sense - will gather the information and report back. Yes, it's via CAN bus directly and Victron supports it out of the box, no additional drivers were installed

petrklus, 2025-05-12:
Here it is: @YetAnotherOneGuru @Dr-Gigavolt 

[image]

In text format:

```
com.victronenergy.battery.socketcan_can1
Alarms/CellImbalance                                   0
Alarms/ChargeBlocked                                   -
Alarms/DischargeBlocked                                -
Alarms/HighCellVoltage                                 0
Alarms/HighChargeCurrent                               0
Alarms/HighChargeTemperature                           0
Alarms/HighDischargeCurrent                            0
Alarms/HighTemperature                                 0
Alarms/HighVoltage                                     -
Alarms/InternalFailure                                 0
Alarms/LowChargeTemperature                            0
Alarms/LowTemperature                                  0
Alarms/LowVoltage                                      0
Capacity                                             294
Connected                                              1
ConnectionInformation                                  -
CustomName
Dc/0/Current                                         9.4
Dc/0/Power                                           501
Dc/0/Temperature                                      28
Dc/0/Voltage                                        53.4
DeviceInstance                                       512
Diagnostics/Module0/Alarms/CellImbalance               -
Diagnostics/Module0/Alarms/HighCellVoltage             -
Diagnostics/Module0/Alarms/HighChargeCurrent           -
Diagnostics/Module0/Alarms/HighChargeTemperature       -
Diagnostics/Module0/Alarms/HighDischargeCurrent        -
Diagnostics/Module0/Alarms/HighTemperature             -
Diagnostics/Module0/Alarms/InternalFailure             -
Diagnostics/Module0/Alarms/LowChargeTemperature        -
Diagnostics/Module0/Alarms/LowTemperature              -
Diagnostics/Module0/Alarms/LowVoltage                  -
Diagnostics/Module0/Id                                 -
Diagnostics/Module1/Alarms/CellImbalance               -
Diagnostics/Module1/Alarms/HighCellVoltage             -
Diagnostics/Module1/Alarms/HighChargeCurrent           -
Diagnostics/Module1/Alarms/HighChargeTemperature       -
Diagnostics/Module1/Alarms/HighDischargeCurrent        -
Diagnostics/Module1/Alarms/HighTemperature             -
Diagnostics/Module1/Alarms/InternalFailure             -
Diagnostics/Module1/Alarms/LowChargeTemperature        -
Diagnostics/Module1/Alarms/LowTemperature              -
Diagnostics/Module1/Alarms/LowVoltage                  -
Diagnostics/Module1/Id                                 -
Diagnostics/Module2/Alarms/CellImbalance               -
Diagnostics/Module2/Alarms/HighCellVoltage             -
Diagnostics/Module2/Alarms/HighChargeCurrent           -
Diagnostics/Module2/Alarms/HighChargeTemperature       -
Diagnostics/Module2/Alarms/HighDischargeCurrent        -
Diagnostics/Module2/Alarms/HighTemperature             -
Diagnostics/Module2/Alarms/InternalFailure             -
Diagnostics/Module2/Alarms/LowChargeTemperature        -
Diagnostics/Module2/Alarms/LowTemperature              -
Diagnostics/Module2/Alarms/LowVoltage                  -
Diagnostics/Module2/Id                                 -
Family                                           PREMIUM
FirmwareVersion                                      282
Id                                                     -
Info/BatteryLowVoltage                                43
Info/ChargeRequest                                     0
Info/FullChargeRequest                                 -
Info/MaxChargeCurrent                                216
Info/MaxChargeVoltage                               58.4
Info/MaxDischargeCurrent                             216
InstalledCapacity                                    300
Manufacturer                                         BYD
Mgmt/Connection                                  CAN-bus
Mgmt/ProcessName                             can-bus-bms
Mgmt/ProcessVersion                                v0.62
ProductId                                          45081
ProductName                       BYD Premium LV battery
Redetect                                               0
Sense/Current                                          -
Sense/Soc                                              -
Sense/Temperature                                      -
Sense/Voltage                                          -
Serial                                                 -
Soc                                                   98
Soh                                                   98
System/MaxCellTemperature                             30
System/MaxCellVoltage                              3.337
System/MaxTemperatureCellId                            1
System/MaxVoltageCellId                                1
System/MinCellTemperature                             28
System/MinCellVoltage                              3.327
System/MinTemperatureCellId                            1
System/MinVoltageCellId                                1
System/NrOfModulesBlockingCharge                       -
System/NrOfModulesBlockingDischarge                    -
System/NrOfModulesOffline                              0
System/NrOfModulesOnline                               1
```

Dr-Gigavolt, 2025-05-12:
Is it all? Voltages of cells not available? In such a case you will have to modify my program. With a bit of Python knowledge you'll manage it in a couple of hours.

petrklus, 2025-05-12:
> Is it all? Voltages of cells not available? In such a case you will have to modify my program. With a bit of Python knowledge you'll manage it in a couple of hours.

Yes, that's all I can see, the individual cells do not seemto be there via can. I know there is a PC diagnostic software that can display the individual cell voltages but it does not seem to be exposed via CAN.

Would this be a problem if we only have the `MinCellVoltage` and `MaxCellVoltage` without the individual cells?

I am pretty competent python coder, where would you recommend I start? Is the software ready for different drivers/modular or would I need to add this functionality?

YetAnotherOneGuru, 2025-05-13:
> Yes, that's all I can see, the individual cells do not seemto be there via can. I know there is a PC diagnostic software that can display the individual cell voltages but it does not seem to be [...]

The reason, why You would prefer the voltages of the single cells is to avoid 1 cell overshooting the rest. With a BYD pack this is probably very unlikely, because the cells will be matched, but 1 faulty cell could cause unpredictable results. So for me it would be a safety aspect to react on issues with single cell voltages also in aggregate batteries. This is why I personally use the serial connection with serialbattery driver and not the can connection also offered from the JKBMS.

Would You invest some time and have a look, if there is a way to get the single cell voltages to the cerbo? 
E.g. maybe byd offers a serial connection as well or maybe there is an option with a different protocol to provide the single cell values? 
E.g. If the BYD BMS is capable of a serial connection, You can buy/manifacture a serial connector - have a look at the videos from offgrid garage on Youtube.

If not, it would need imho a separate module, that reacts on devices like com.victronenergy.battery.socketcan_can1 and can handle batteries without single cell data. Since handling batteries is the main function of the aggregate_batteries, it is imho more or less a duplication of code of this main function (minus the single cell voltage calculations of course, but still).

Dr-Gigavolt, 2025-05-14:
My program won't accept BYD because it stops if at least one required value is missing. But you can modify it - remove everything which works with the individual voltages and use what is available. As the minimum and maximum cell voltages are transmitted the safety features can be maintained. Therefore even without additional hardware you could aggregate CAN and multiple serials.

Dr-Gigavolt, 2025-05-14:
And think about the different series resistances of different batteries. If there is a coarse difference of current splitting in respect to the capacity ratio you need to add some cable length to the battery(ies) getting too much current.

mr-manuel, 2025-09-29:
For BYD you can read the cell voltages via network and a software, but I have no clue about the protocol used.

Single cell voltages, single temperature values, balancing status and some other are a feature of dbus-serialbattery. Official Victron list can be found here: https://github.com/victronenergy/venus/wiki/dbus#battery

Shouldn't be that huge problem, if you only have min and max cell voltage. From the battery voltage you can calculate the average cell voltage and if min or max cell voltage has a big delta, then it's probably one cell spiking. It would also be possible to implement in the code, that only available values from all batteries are used, but that is more a design decision of the developer.

Else there are other possibilities to aggregate your batteries:
* https://github.com/pulquero/BatteryAggregator
* https://github.com/mr-manuel/venus-os_dbus-mqtt-battery you have to aggregate the data via Node-RED and then send it via MQTT to the driver

I'm planning to add a MQTT source to dbus-serialbattery in Q4 2025, so you can use any battery with dbus-serialbattery, if you have the data somewhere.