Dr-Gigavolt/dbus-aggregate-batteries #103: Aggregation stops after a while

gwiddison, 2025-08-10:
I have the following setup - 2 x JKBMS B2A8S30P and am using a Raspberry Pi with Cerbo GX talking over bluetooth, all seems ok for a while and I get aggregation for a few hours then it stops seeing the batteries, I can still see the individual batteries but not the aggregate one, if I look through the logs the aggregate battery can then see either one or zero batteries, if it is one that soon goes to zero, if I use DBUS-SPY at this point I only have about 10 lines instead of the pages worth that I usually get, rebooting the Pi usually sorts this out for a while.
I am guessing that DBUS stops "seeing" the batteries properly but do not have a clue where to start looking myself!

gwiddison, 2025-08-10:
[image]

Dr-Gigavolt, 2025-08-13:
Hello, if even the dbus-spy does not work properly, the issue is outside of my program. I don't no why, I'm not a software expert. But as a hardware engineer I recommend avoid wireless communication everywhere where reliability is required. Perhaps this could solve your problem.

Dr-Gigavolt, 2025-08-13:
I am happy that your system works. It is confirmed again - keep it simple, than it works :-). I hope you ordered serial adapters with galvanic insulation. It is necessary, otherwise in case of BMS disconnection blows something up.

closed 2025-08-13 by Dr-Gigavolt

gwiddison, 2025-08-13:
Hi, I used adaptors that are sold to go between the JKBMS and the RS485 to USB adaptor, it doesn't state explicitly that they are galvanic isolators but I am assuming (hoping? :-)) that they are!
They are made for JKBMS's though.