Dr-Gigavolt/dbus-aggregate-batteries #118: Failure "Number of cells of batteries is not correct. Exiting"

Chris87-ux, 2025-09-17:
Venus OS 3.66
Newest SerialBattery (Installed yesterday)
Newest dbus_aggregate (Installed yesterday)

Batteries are shown correctly:
[image]
BMS Set to 14 Cells and also dbus_syp displays on every BMS 14 cells.

Settings also modified to 14cells
[image]

Failure Log:
[image]

Also had some issues with the existing instruction for installation, the dbus-aggregate-batteries wouldn't start, just abter a manual Start I got it to display the instance.

Help would be very much appreciate.

/Chris

Dr-Gigavolt, 2025-09-17:
Hi Chris,

What do you mean "newest dbus_aggregate?" Have you installed the release, or cloned the last commit from the repo?

We have troubles with the new Venus OS, see my newest comment in readme.md, there is also a proposal of temporary solution. 

I deleted one release too much few days ago, now it is back. The 3.5.20250707 is the one I still have on my system and it runs reliably with Venus 3.55. 

The error with number of cells is mysterious. If you installed the release from June 2025 and it happens it could be caused by the incompatibility with the new Venus OS. If you have cloned the latest commits I will ask Andreas to double-check his contribution.

Chris87-ux, 2025-09-18:
Hi @Dr-Gigavolt, 
thanks for you reply.
I used the installation script as mentioned: 
wget -O /tmp/install.sh https://raw.githubusercontent.com/Dr-Gigavolt/dbus-aggregate-batteries/main/install.sh

bash /tmp/install.sh

My installation is not time sensitive, so in case you need someone for test or so on feel free to provide some instructions. 
If it is important, my installation includes 4 JK_BMS with each 14cells of 28P 18650 Li-Ion batteries.

Really looking dforward to test your code.

/Chris

Dr-Gigavolt, 2025-09-18:
Just install the latest release as you did last time.

Chris87-ux, 2025-09-18:
Thanks,
unfortunately still the same:

[image]

tobio-de, 2025-09-20:
Hi there! I have the same issue. 2 x JKBMS 17s 
NR_OF_CELLS_PER_BATTERY = 17

INFO:root:Wed Sep 17 11:40:14 2025: SerialBattery(JKBMS) found, named as: JK_B2A20S20P.

ERROR:root:Wed Sep 17 11:40:14 2025: Number of cells of batteries is not correct. Exiting.

Version 3.5.20250516 on VenusOS 3.66 (Cerbo)

tobio-de, 2025-09-20:
Now I updated to 3.5.20250707
still the same result:
INFO:root:Wed Sep 17 11:40:14 2025: SerialBattery(JKBMS) found, named as: JK_B2A20S20P.
ERROR:root:Wed Sep 17 11:40:14 2025: Number of cells of batteries is not correct. Exiting.